polyend/TrackerBetaTesting #1453: dummy track for extra effects

8theta, 2022-04-05:
Hei, the idea is simple:
-imagine that you have a sequence as usual, in track 1. 
-in track 2 you could choose an instrument that is called "track 1"
-This way track 2 will have no notes, but insead can send additional effects to track 1

I cannot remember which clasic tracker had this function, but the closest i can think of in current instrument sequencers is the free allocation of LFOs in the elektron machinedrum.

cheers,
Themis
thetasounds.org

theorize999, 2022-04-08:
I wouldn't call it a classic tracker, even though it's very old, but SunVox works like that, it's a great system but there's unlimited tracks... because of the track limit I wouldn't use this feature much for composing, but for sound design I could see it being quite handy.

PITmast, 2022-04-13:
Hi

There is already such a functionality, in Tracker, at least for midi. Instead of T1-T8 use the same instrument number in another track. But I didn't test it for samples to be honest.

Piotr

PITmast, 2022-04-13:
My mistake, it'a available for midi only. You can delete note having Mxx instrument. For samples it deletes everything.

Piotr