polyend/TrackerBetaTesting #1458: [1.5 & 1.6.0b1] Clicking on sample start when using filter FX commands

MattWilsh, 2022-04-14:
When using the filter FX commands on samples, I'm getting a lot of clicking on the sample start when playing in context of the tune.

Previewing the step doesn't reproduce it. Playing the sample alone doesn't reproduce it.

### Steps to reproduce

Using the filter FX command, low pass and band pass (it may do it on high pass too) on an individual line after another sample recreated it. Previewing the line with it on doesn't reproduce it. 

Interestingly on one of the videos I've replaced the sample on Line 1 to the same sample. And replaying the track before the last sample has finished, the click happens _without_ any filter FX command but it does follow the sample playing with a filter FX command.

### Other important information

* Bug happens every time, on any project (I use a lot of filter FX on breaks & bass to create variation)
* Does this bug happen in previous firmware versions? - Yes
* Does this bug happen in every project? - Yes

MattWilsh, 2022-04-14:
Videos here :)

https://youtu.be/SP8r6laiBlg

https://youtube.com/shorts/-dEuGoZIV40

https://youtube.com/shorts/YD6v6FhesWM

breskar, 2022-04-27:
In the first example you are going from sample 19 to 18 and I think that the click happens when the 19 stops.
Try putting an OFF command in place of the second bass bass note and it should still click.
In this case I would try to change the Instrument volume release parameter or put an  OFF / FAD command somewhere before the second bass note.

MattWilsh, 2022-05-02:
Thanks for your response. Sadly makes no difference. I started a new project last night and the exact same happens, I’ve tried CUT, OFF and FAD to see if there’s any different behaviour between them. Taking the filter FX command off seems to reduce the amount of clicking, so feels like the code around that and using volume FX cause the clicks the most.

PRAYTHEPO, 2022-05-04:
1.6.0b4 still the same problem. Especially obvious when you use Volume FX with granular synthesis. Volume jumps from 80 to 15 responds with clicks every time

sebpmtl, 2022-05-18:
I noticed that problem. Clicking sounds everywhere with LFOs and filters! But also with the volume too. As soon as you start automating stuff, you get clicks. I'm often able to cope and make it work but it doesn't work as it should.

PRAYTHEPO, 2022-05-30:
Also got clicking while using ARP FX. Happens everytime, ADSR parameters won't help

notaprogrammer9, 2022-06-18:
Clicks everywhere. 1.3.0 didn’t click as much. Please fix

doelman, 2022-07-28:
Dear Polyend, could you please please let us know when all those issues that causes clicks and artifacts are going to be solved?

The current firmware (even though many new good features were added along the way) is just not sufficient for making proper music, it's highly frustrated to work around all the issues and I think they've been addressed many times now.

Not sure if anyone mentioned this but when you're using different samples on the same track (just after each other) clicks / loud pops occur all the time.

I love the device but at the moment it's killing my creativity.

notaprogrammer9, 2022-07-28:
Yep it’s terrible. 

I have bought one about a year ago, sold it after two weeks. Bought another one a month ago, noticed the insane clicking and clipping and sold it after 2 days. Not gonna buy a third one. When a unit says “Polyend” I’m skipping. And I will advice everybody I know against buying anything from Polyend. 

These devs pull and mark and issue as “fixed” but in reality not much gets fixed. I have wasted 150€ on these Trackers. These are fraudulent practices.

doelman, 2022-10-11:
Yes this problem is so super annoying and makes the Tracker unusable for
me. Can't believe that this didn't pop up during testing on Polyends side.
BTW you can avoid it by having an OFF command between the two samples. But
this of course is not possible when you want the other sample directly on
the next step.

Grapes-HH, 2022-11-29:
As much as i love the device, the clicks on volume fx make it unusable as a sampler and i only use it as a sequencer right now. if this does not get fixed soon i will sell the tracker and turn my back on polyend as a costumer. i already advise all my producer friends against the tracker right now. better go renoise or M8 for sure.

PRAYTHEPO, 2022-11-30:
> As much as i love the device, the clicks on volume fx make it unusable as a sampler and i only use it as a sequencer right now. if this does not get fixed soon i will sell the tracker and turn my [...]

@Grapes-HH 
You still think they care? From release of 1.6 firmware and Polyend Play in July they don't care about their customers anymore. Poland FPS from engineering world. They got their profit from naive customers and nothing else bothers them

breskar, 2022-11-30:
Such a shame this could have been such a better device.

closed 2023-04-21 by MiroPoly