polyend/TrackerBetaTesting #1462: Allow Step FX to be applied without assigning a sample

tfspsound, 2022-04-19:
What I’d like to suggest is that we could apply other fx to a triggered sample without retriggering the sample. 
So for example if a roll is used, 4 steps down another roll setting could be applied changing it. 
Another example might be if a long sample is playing, a tune fx could be applied changing its pitch halfway through or multiple times. 
Basically once a sample is triggered any fx applied on the same track effects the sample that is playing. 
It’s been a while but I think I remember being able to do something similar on LSDJ but not with samples.

bambrose, 2022-04-19:
This would be amazing, and extend the functionality significantly if possible... not re-triggering the envelope, but still retriggering/adjust the associated effect (I think a 'trig lock' in Elektron parlance) would be a significant enhancement. I suppose probably not possible, but I second this request

PRAYTHEPO, 2022-04-29:
It's already possible if i understand correctly what you want

https://user-images.githubusercontent.com/58429366/165937793-7e7396cf-b04b-40c0-b08b-87345f8ade88.MOV

tfspsound, 2022-05-03:
@PRAYTHEPO You’re right! When I tried it prior to posting this I was trying to change roll speed and it didn’t work. Seems to now.

closed 2022-05-03 by tfspsound