polyend/TrackerBetaTesting #1465: Jog wheel lifespan / alternative

tfspsound, 2022-04-21:
I’ve just had to clean my jog wheel for the second time hoping this will give me a little bit of life until the next batch of encoders reach the ‘shops’. I don’t know how many times I will be able to bend those pins back before they will snap off and not being handy with electronics I don’t fancy my chances of being able to replace the encoder fully. 
I worry that at some point I may end up with a tracker that it takes forever to edit on. 
Its good that we can use the Up and Down keys to adjust parameters but it would be even better if we could use it in conjunction with Shift to move the cursor in greater increments like 1/10 of a second, or since the screen keys allow us to select a parameter to adjust why not use Left and Right to adjust which decimal place we adjust?
I hope you will take this into consideration and think about how you could minimise jog wheel use to increase its longevity as I’ve only owned it for a year and a half and already the encoder is shot. 

Cheers

Jacknametrouble, 2022-04-21:
This is a hardware-related issue, please reach out to https://help.polyend.com/submit_ticket and our customer support team will help you.

closed 2022-04-21 by Jacknametrouble

tfspsound, 2022-04-22:
Hey @Jacknametrouble 
I will reach out to customer services but with all due respect I think it’s a hardware issue that could be lessened by some software support. 
I also noticed that when editing something I had recorded in, that the Left and Right buttons didn’t the effect Start or End points in the Sample Editor as they do on the Sample Playback screen, which would be nice.
I’m only coming from a place of looking after the tracker in years to come. Besides you guys don’t manufacture the encoders so it isn’t a slight on the build quality of the tracker. 

Cheers

trutypesounds, 2022-04-22:
> Hey @Jacknametrouble I will reach out to customer services but with all due respect I think it’s a hardware issue that could be lessened by some software support. I also noticed that when editing [...]

i've had the same proposal somewhere here on github about using shift+arrow keys, but there was no proper response on this issue. im near to give up on this machine, not sure yet, because its such a nice idea and layout but performed very badly mostly on all levels, i wish they did a collab with elektron or te on this, or just plain elektron or te released a hardware tracker done right, but not sure they are interested in this... its a bit strange btw, because trackers come from demoscene and swedes are the most advanced and active at this field... since Renoise released a raspberry pi build (and apple arm) im thinking of switching to this side, but its a tricky one, because you cant pack it all in a nice and neat package in this case and be portable.