polyend/TrackerBetaTesting #1467: Microtonal support, load .tun .scl files which automatically add the correct fine tune FX for the corrosponding notes.

stekkermane, 2022-04-22:
### A new feature

Like OpenMPT, renoise and other modern trackers they have support for microtonal scales to be loaded,
I think this would be an amazing feature to also add into the polyend tracker.

Right now i have to manually lookup each finetune setting for the specific note using software on my computer and add the finetune step FX myself by hand.
this is really slow and painfull to do, With the option to just load a tuning scale in the config menu just like how u can do that for 12 TET scales like minor/major/phrygian etc.

Plieuw, 2022-05-11:
This would be great

Undistraction, 2022-12-21:
As an additional request, it would be nice to allow fills to draw from not just a scale, but a user-defined selection of notes (possibly spanning multiple octaves).

tobia, 2023-12-16:
I just came in here to say that, for full microtonal support, the tuning files should be able to specify **_the entire list of notes,_** which can be much more that 128 and do not necessarily repeat by octave; as well as one (or more?) **_pad layouts_** that map pads onto notes.

Additionally, the tuning should specify what happens when the user hits Shift+Jog to adjust a note by "octave", since alternative tunings do not necessarily repeat by octave, nor by 12 steps. Same for adjusting the pad layout "octave".

A few examples follow.

Taking this as the standard 12edo MIDI scale:

```
Name     Cents  MIDI Note  MIDI Bend
-----  -------  ---------  ---------
C0        0.00  (C0)   12          0
C#0     100.00  (C#0)  13          0
D0      200.00  (D0)   14          0
D#0     300.00  (D#0)  15          0
E0      400.00  (E0)   16          0
F0      500.00  (F0)   17          0
F#0     600.00  (F#0)  18          0
G0      700.00  (G0)   19          0
G#0     800.00  (G#0)  20          0
A0      900.00  (A0)   21          0
A#0    1000.00  (A#0)  22          0
B0     1100.00  (B0)   23          0
C1     1200.00  (C1)   24          0
```

One may want to use a [Bohlen–Pierce EQ scale](https://en.wikipedia.org/wiki/Bohlen%E2%80%93Pierce_scale) which has the following notes:

```

Name     Cents  MIDI Note  MIDI Bend
-----  -------  ---------  ---------
C0        0.00  (C0)   12          0
C#0     146.30  (C#0)  13       1896
D0      292.61  (D#0)  15       -303
E0      438.91  (E0)   16       1594
F0      585.22  (F#0)  18       -605
F#0     731.52  (G0)   19       1291
G0      877.83  (A0)   21       -908
H0     1024.13  (A#0)  22        988
H#0    1170.44  (C1)   24      -1211
J0     1316.74  (C#1)  25        686
A0     1463.05  (D#1)  27      -1513
A#0    1609.35  (E1)   28        383
B0     1755.66  (F#1)  30      -1816
C1     1901.96  (G1)   31         80
```

Notice this scale has names like "H" and "J" and it does **_not_** repeat by octave. If we start this scale's "C0" with MIDI Note (C0), then its "C1" is close to MIDI (G1), not (C1).

Another example, a [19 equal temperament scale](https://en.wikipedia.org/wiki/19_equal_temperament) has the following notes:

```
Name     Cents  MIDI Note  MIDI Bend
-----  -------  ---------  ---------
C-0       0.00  (C0)   12          0
C#0      63.16  (C#0)  13      -1509
Db0     126.32  (C#0)  13       1078
D-0     189.48  (D0)   14       -431
D#0     252.64  (D#0)  15      -1940
Eb0     315.80  (D#0)  15        647
E-0     378.96  (E0)   16       -862
E#0     442.12  (E0)   16       1725
F-0     505.28  (F0)   17        216
F#0     568.44  (F#0)  18      -1293
Gb0     631.60  (F#0)  18       1294
G-0     694.76  (G0)   19       -215
G#0     757.92  (G#0)  20      -1724
Ab0     821.08  (G#0)  20        863
A-0     884.24  (A0)   21       -646
A#0     947.40  (A0)   21       1942
Bb0    1010.56  (A#0)  22        433
B-0    1073.72  (B0)   23      -1076
B#0    1136.88  (B0)   23       1511
C-1    1200.00  (C1)   24          0
```

Notice in this case the scale does repeat by octave, but it has more than 12 notes per octave, because most pairs like "C#" and "Db" are distinct.

It gets even more complicated with scales such as [53 equal temperament](https://en.wikipedia.org/wiki/53_equal_temperament), where not only G# ≠ Ab, but there are 10 notes just between G and A, with different naming conventions that get harder and harder to squeeze into 2 or 3 ASCII characters:

```
Diatonic:       G  G#  GX   GX#  GXX  ABB  ABb  AB   Ab  A
Pythagorean:    G  FX  EX#  BBb  Ab   G#   FX#  CBb  BB  A
Ups and downs:  G  G^  G^^  Abv  Ab   G#   G#^  Avv  Av  A
```

Finally, I should note that a great resource to generate and inspect scales is Sevish's [Scale Workshop](https://sevish.com/scaleworkshop/).