polyend/TrackerBetaTesting #1469: automatic tune sample to fit pattern length

gherkins, 2022-04-23:
<sup>searched for it, didn't find it, still think it's a good idea...</sup>

Imagine this:

* go to instrument parameters
* hold down shift 
* "tune" becomes "tune to pattern length"
* click and have the tuning/finetuning settings adjusted to make the sample fit the pattern length

*instant seamless looping* 🎉

Now that would be really nice, wouldn't it?

breskar, 2022-04-26:
Yes please, the good old sync sample length to a number of steps.

theorize999, 2022-04-28:
please please please introduce this, It really is a pain in the rear doing this manually, and I'd rather see this fit to xx steps than pattern length personally since xx steps could be set to pattern length. This would be a huge improvement!

breskar, 2022-04-28:
as #1466  Sample Tune to Lines

Djeroek, 2022-05-01:
+1 Would be great to have syncing options to X amount of lines for samples, auto transposing and finetuning until it fits.

the-invisible-bears, 2022-05-02:
syncing sample length to pattern length would be a godsend.  I use it all the time in my software tracker and it's the one feature I was a little bummed to find out didn't exist on the Polyend Tracker. Please, bring this!

Beddings, 2022-06-11:
Yes, please, a thousand times.

NoSitRecords, 2022-06-11:
Yesssssssss

shawnroos, 2022-06-20:
➕ 💯