polyend/TrackerBetaTesting #1471: EQ values problems and new features suggestions

PRAYTHEPO, 2022-04-25:
It was mentioned before in some threads, but EQ in PT is very subtle and does not correspond with its real world values: in sample editor when i bring down at 500 range for around 3db i cant hear the difference. For 3db i expect significant changes to the sound, but that doesn't happen.
Almost the same happens with Master EQ: 300 and 900 -3db - almost no difference. 

My suggestions are:
- match EQ gains to the real world values
- limit decreasing and increasing ranges to maybe 5db
- make more smooth changes with jogwheel to the freq gain
- add low/high cut filter to the sample editor

- - - - - 

[ P.S. i know about intentional limitations in PT, but(!) we are not in 1991, we are 30 years later. I want to produce music on this device which meets modern requirements such as good EQ'ing and stereo sampling.

I know how angry you are about stereo, but i think that option to choose between vintage mono behavior and modern stereo will sell this device to the younger generation, will make PT more user friendly for a newcomers who are not familiar with trackers at all. In my case i want to write some synthwave with lush and big pads and "doomlike" metal on PT with samples of my guitar in stereo so that i use 1 track instead of 2 when mono samples are used

PRAYTHEPO, 2022-04-25:
Forgot to mention: working on 1.6.0b1 firmware

[EQ works much better in 1.5.0, tested just a minute ago]

Djeroek, 2022-11-06:
yeah, the EQ doesn't work here in 1.6 when I try to process/render in changes into the sample. I tried removing sub frequencies in samples with no noticeable change.