polyend/TrackerBetaTesting #1476: [1.6.0b2] Bug title

PITmast, 2022-04-29:
Guys,

Please check Jog wheel turn left slowly click-by-click in project list or in pattern view for instance.
It's no encoder hardware issue. It never happened with previous firmwares.

It's hard to comment.

regards
Piotr

Synthdawg, 2022-04-29:
Seeing same thing. Rotating counter clockwise has inconsistent clicks per step change. Clockwise ok, 1 step per click.

PRAYTHEPO, 2022-04-29:
Counter-clockwise not consistent. Clockwise is ok

Jacknametrouble, 2022-04-29:
@PITmast nice catch, fix coming in a moment. Thank you!

bambrose, 2022-04-29:
Yeah good spot. This is across all parameters as far as I can see (I.e even on stuff like panning / volume etc)
No counter clockwise functionality is correct 
Thank goodness someone else noticed this… it would have driven me crazy trying to work out what was wrong

Jacknametrouble, 2022-04-29:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b3

closed 2022-04-29 by Jacknametrouble

bambrose, 2022-04-29:
Bloody he’ll that was quick. Polyend never sleeps obviously 
Nice one ☝️