polyend/TrackerBetaTesting #1477: Performance mode eternal scrolling FX type and total freeze

PRAYTHEPO, 2022-04-29:
After changing performance mode fx for about 5 times it began to scroll eternally and now PT is totally frozen. Only reset helps

https://user-images.githubusercontent.com/58429366/166064084-2d249009-8031-4ae2-9f3e-2809bbf8d841.MOV

Jacknametrouble, 2022-04-29:
Hey, just spent the last two hours trying to reproduce and looking in the code, with no success. Any chance an arrow key got stuck in your unit?

PRAYTHEPO, 2022-04-30:
> Hey, just spent the last two hours trying to reproduce and looking in the code, with no success. Any chance an arrow key got stuck in your unit?

I was clicking arrows while this happened, no effect. I cannot reproduce it too, but i will keep it in mind

trutypesounds, 2022-04-30:
@PRAYTHEPO @Jacknametrouble  i just had this while was working on a tune

https://user-images.githubusercontent.com/17484006/166106483-81f4a72f-41be-4dd4-be1f-d7d0e1d3f691.MOV

tracker just started to roll backwards with no sound going on, no button functions except the grid i still can play an instrument...
i just hope that tracker autosaved my stuff....

PRAYTHEPO, 2022-04-30:
The same thing just happened when i was playing NES games. Left arrow button got stuck (in software) and my character was eternally moving left without responding to other buttons. It's some software issues, i think so. Only reboot helps

trutypesounds, 2022-04-30:
> The same thing just happened when i was playing NES games. Left arrow button got stuck (in software) and my character was eternally moving left without responding to other buttons. It's some software issues, i think so. Only reboot helps
> 
> (As i can see, this bug affects only arrow keys)

i guess not arrow keys affected, but its a random, that some keys are get stuck/sticky and nothing helps.

https://user-images.githubusercontent.com/17484006/166113749-5f47739c-f09e-470a-8fe9-d829ed32899c.MOV

PRAYTHEPO, 2022-05-01:
New freeze type: i was in Instrument Parameters when 48 pads stopped to response. But i can still go to other menus

trutypesounds, 2022-05-02:
> New freeze type: i was in Instrument Parameters when 48 pads stopped to response. But i can still go to other menus

@Jacknametrouble @PRAYTHEPO same happened on pattern screen today twice in 2 hours, also one freeze happened after i turned the pt on and was messing with waveforms and granular, the whole tracker freezed

trutypesounds, 2022-05-02:
just rebooted went to song mode - all buttons stopped working except the grid, still can play an instrument, pls guys at polyend check this asap

trutypesounds, 2022-05-02:
well, something new just happened - i loaded a new wave into empty sample slot but it also applied to the previous sample...
@Jacknametrouble @wjakobczyk-polyend

Jacknametrouble, 2022-05-03:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b4

closed 2022-05-03 by Jacknametrouble