polyend/TrackerBetaTesting #1480: PT freezes in Sample rRcorder (Radio) [1.6.0b3]

PRAYTHEPO, 2022-04-30:
I was trying to tune in some radio when my PT suddenly frozen. Then i tried again, everything was working fine until it suddenly frozen again. I didn't do anything. Just power on PT -> go to sample recording -> tune with jogwheel.
First time it frozen immediately, second time after ~2 minutes.

https://user-images.githubusercontent.com/58429366/166115233-5b6ece3c-93c3-4ec1-900d-2d3b1da38312.MOV

trutypesounds, 2022-04-30:
@PRAYTHEPO i tried to replicate this with no luck, but i must admit this b3 is full of that freeze/sticky keys issues appearing randomly from time to time =/

Jacknametrouble, 2022-04-30:
I cannot get any freeze/sticky keys. @PRAYTHEPO How are you powering your Tracker unit? Use an alternative powering source, and check other cables.

PRAYTHEPO, 2022-05-01:
@Jacknametrouble I am using original cable and power adaptor. And as you can see, @trutypesounds also gets some freezes.

rollthestone, 2022-05-01:
@PRAYTHEPO same here. Freezes also occur during granular synth parameteres mangling.

trutypesounds, 2022-05-01:
> @PRAYTHEPO same here. Freezes also occur during granular synth parameteres mangling.

i had this too once

trutypesounds, 2022-05-01:
> @Jacknametrouble I am using original cable and power adaptor. And as you can see, @trutypesounds also gets some freezes.

@PRAYTHEPO @Jacknametrouble mine powered via powerbank mostly, rarely via pt power adapter, but there was no freezes/sticky keys issues in 1.6.0b1

trutypesounds, 2022-05-01:
> I cannot get any freeze/sticky keys. @PRAYTHEPO How are you powering your Tracker unit? Use an alternative powering source, and check other cables.

ppl on pt discord reports freezes/weird behaviors too

Jacknametrouble, 2022-05-03:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b4

Jacknametrouble, 2022-05-03:
@trutypesounds please drop at Discord if possible, thank you!

closed 2022-05-03 by Jacknametrouble

trutypesounds, 2022-05-04:
@Jacknametrouble im not happy to report - i just had a total freeze at sample recording screen while trying to sample something off the built in radio (pt powered from powerbank as i usually do)

trutypesounds, 2022-05-04:
@Jacknametrouble just had a second total freeze in sample screen