polyend/TrackerBetaTesting #1484: Ability to Sort Projects in File Menu (by date modified? or by name?)

teamoth, 2022-05-02:
Please can we have a way to sort projects in the File menu? They don't seem to be sorted by anything (as far as I've been able to figure out), not alphabetical and not date created/modified. Newly created projects don't appear at the top and it can be hard to find them, as I tend to use the auto-naming feature and don't remember the obscure names always!

Also, when I look at the projects folder on my computer it says that they were all created on 31/12/2017. Not that it's an issue for me though.

dustboxednorth, 2022-05-04:
> Also, when I look at the projects folder on my computer it says that they were all created on 31/12/2017. Not that it's an issue for me though.

This is probably unrelated, but to give you a little bit of closure on this, keep in mind that the tracker doesn't have any realtime clock hardware (at least not that I know of), so doing this wouldn't be possible, at least not in any conventional sense. Something something, using radio broadcasts to obtain time signals? I have no clue, honestly. Sounds unlikely to be able to pull off.

e10k, 2022-05-21:
I believe for most users the following improvements would make the projects browser much more usable:

- alphabetical ordering
- ability to use subfolders (1 level deep should be enough)
- ability to rename projects

With these, problems related to not having a realtime clock would become non-issues, at least as far as the projects browser is concerned.

Related: https://github.com/polyend/TrackerBetaTesting/issues/1095

teamoth, 2022-05-21:
Alphabetical ordering and subfolder support would be great.

Renaming projects is already available on the Tracker I believe. I'll recheck this later today.

teamoth, 2022-08-03:
I've been doing a bit of organising my Tracker projects recently and noticed that you can't actually rename projects (I'm not sure why I thought you could, sorry), but to get around this you can load a project, use the 'save as' option, create a new copy and delete the old one. 

I still find it problematic that the sorting isn't alphabetical, or numerical. I renamed a load of projects last night so that they started with 01 and only some of the recently created ones appeared at the top of the list. 

I wonder if perhaps it is sorting in a strange way based on the version of the firmware that I was using when the project was first created? I've had probably all the firmwares and betas on my machine and the project sorting seems to roughly put older stuff from older firmwares at the bottom.