polyend/TrackerBetaTesting #1486: [1.6.0b3] Bug -> locked pattern + length (video)

anemlife, 2022-05-02:
Hello, 

Just started using 1.6.0b3 yesterday and had a really strange bug after 5 minutes.

https://www.youtube.com/watch?v=bEHuKg5awjE&feature=youtu.be

Basically I suddenly got locked in pattern 1. No way to switch to another pattern (tried with pads/jogwheel). Same with the pattern length, it was locked at 4. Can't tell what caused it. Had to reboot the machine and it worked fine again.

(Also had another bug where I got locked in pattern view, no way to switch to another screen, no way to type any note. But unfortunately I did power off the machine before catching it on video so I'm not opening another ticket for this one)

Switching back to 1.5 for now

PRAYTHEPO, 2022-05-02:
Looks like this #1477

https://github.com/polyend/TrackerBetaTesting/issues/1477#issuecomment-1114170528

Jacknametrouble, 2022-05-03:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b4

closed 2022-05-03 by Jacknametrouble