polyend/TrackerBetaTesting #1495: Renaming/Labeling instruments

0x7CA, 2022-05-04:
It would be very useful to be able to rename or label the instruments in the instrument list. Especially when working with external MIDI gear, you need to memorize which channel is the kick, closed hat, open hat etc.
Something like the track renaming but then triggered by pressing rec on the hovered instrument in the instruments list.

arphex, 2022-05-05:
this could also go for renaming midi cc's from the instrument menu if possible - i presume it will not be much overhead memory wise (same number of characters)

mrmarbury, 2022-07-29:
Yes please!

Monsieur36, 2022-10-28:
MIDI channel/track renaming/labelling would be very welcome indeed. As would the option to locate MIDI instrument outputs within the main 48 instruments (for easier access and data entry).

danapeks, 2023-01-03:
100% - this would be so helpful