polyend/TrackerBetaTesting #1500: Chord layout for the pads

bycym, 2022-05-06:
Dear devs, is it possible to add a chord layout to the pads? The same one that medusa using. If it could be changed like in the Medusa it would be stunning! 

So if the octaves dimmed in different modes. Thanks a lot!
[image]

bycym, 2022-05-06:
I know, I mean a chord layout that looks like the same in the medusa. For example:
1. Piano keyboard like:
[image]

2. With different pad scale
[image]

Sometimes when I play on the keyboard I cannot see which one is the right pad for the wondered key.

jrfondren, 2022-05-06:
What I do is set the "Pad layout" to the number of notes in the scale. That causes the root note at different octaves to line up at the left edge of the keyboard, e.g., C2, C3, C4, C5 from bottom-left to top-left. You can change the bottom-left note to change which four octaves you have, and you can reach other octaves by starting with the nearest one and using the jog wheel.

This works if you just want to find notes easily with the current firmware on the Tracker, and is similar to one of the Medusa layouts. ... it's also the default, just with the Chromatic scale and a "Pad layout" of 12. If you change the scale you have to also change the pad payout to keep the root notes on the left edge.

Another way to read your request, as you reference "chord layout", is that you want the Tracker to offer an [isometric keyboard](https://www.youtube.com/watch?v=ZczraF3dzU0) layout, where you can press a particular shape of pads to strike a particular chord, and then transpose that chord just by moving your hand and pressing the same shape of pads elsewhere. I'm pretty sure you get that on the Tracker with Chromatic and a "Pad layout" of 5, and then the missing feature is that root notes aren't lit up on the keyboard so it's challenging to navigate.

Jacknametrouble, 2022-05-20:
Thank you for sharing your idea.

closed 2022-05-22 by Jacknametrouble