polyend/TrackerBetaTesting #1501: **Timestretch Beat** leads to compression artifacts and broken rhythm [1.6.0b4]

PRAYTHEPO, 2022-05-06:
I don't know how to explain this and how to reproduce it with other samples. This happens not every time, but sometimes the result is similar

In the video i show result *after*, then *before* cuz of long timestretching process.

https://user-images.githubusercontent.com/58429366/167173605-de487362-8b41-405a-8561-1ed8ca1c4d23.MOV

MattWilsh, 2022-05-30:
Try changing the steps to 64. That’s a 4 bar beat, so it’s potentially trying to stretch it to double speed but in the same about of bars.

PRAYTHEPO, 2022-05-30:
@MattWilsh 
What a problem to speed up beat 2x? This beat is 64, yes, but i don't see any problem to stretch it into 32 steps. Maybe i don't understand how this process works

closed 2024-03-07 by MiroPoly