polyend/TrackerBetaTesting #1502: [1.6.0b3] Fill FX only works on FX2, fills blank on FX1

Linkandzelda, 2022-05-07:
### Steps to reproduce

Select an area, switch to FX1, try to fix some FX. It won't fill out anything in From-To mode.
Switch to FX2 and it fills no problem.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every time

* Does this bug happen in previous firmware versions?
Untested

* Does this bug happen in every project?
Yes, new and old

Jacknametrouble, 2022-05-07:
Hi, thank you for your feedback, any chance for a video clip illustrating this?

Linkandzelda, 2022-05-07:
I went to record the video today, and now I cannot fill even on FX1 and FX2 any type of FX. Am I doing this right?

1. Select an area to fill (whole pattern, couple of rows)
2. Make sure I'm in REC mode
3. Press Fill screen button, to bring up the Fill menu
4. Select FX, select the type of FX, then chose the type of Fill, configure the fill parameters
5. Press Fill on the 8th screen button
6. FX should be filled in, but there's nothing

https://anonfiles.com/hdhbZdd5y6/20220507_131825_mp4

jrfondren, 2022-05-08:
@Linkandzelda you're doing it wrong. The first column of the Fill feature decides which cells will be filled in. Your video has "Fx" as the "Where?", and you apply it to a column without any FX, so the answer to "Where?" is "Nowhere". Try it with "Each" or "Note".

closed 2022-05-08 by Jacknametrouble

Linkandzelda, 2022-05-09:
> @Linkandzelda you're doing it wrong. The first column of the Fill feature decides which cells will be filled in. Your video has "Fx" as the "Where?", and you apply it to a column without any FX, so the answer to "Where?" is "Nowhere". Try it with "Each" or "Note".
> 
> The part you "Select FX" is actually before you press Fill. You select FX by being in an FX column.

Thanks, it works for "No FX".