polyend/TrackerBetaTesting #1507: Faster pattern change in Song mode

PRAYTHEPO, 2022-05-10:
For example if **slot 1** has **pattern 1** and **slot 2** has **pattern 30**, it is really painful to click "next pattern button" 29 times. What if we could use this scheme:

1. Press REC, enter red mode
2. Hold shift while on chosen slot
3. Spin the jogwheel

This would help arrange and rearrange song a lot

trutypesounds, 2022-05-10:
Mentined this too as enhancement, still not implemented. Kills workflow alot.

tfspsound, 2022-05-13:
Would it be possible for Shift to work in conjunction with the screen keys shifting the pattern values in increments of 10 patterns as well?

Synthdawg, 2022-05-13:
> Would it be possible for Shift to work in conjunction with the screen keys shifting the pattern values in increments of 10 patterns as well?

I nearly typed exactly the same thing.
I was thinking:-
Screen Keys +/- 1
Arrow Keys +/- 10
Shift + Jog Scroll through all incrementally

mrmarbury, 2022-07-29:
It's so intuitive to use the jog wheel for this that I constantly reach for it in song mode anyway and mess up things or get completely thrown out of workflow. I usually sort variations of patterns in an Ableton-like grid and when you I the 5th variation of pattern 8 and then the 2nd variation of pattern 1 it is just annoying to dial in especially if I have to do it over and over again.