polyend/TrackerBetaTesting #1516: Dedicated Export Function for Polyend Tracker Instruments

static-noise-PT, 2022-05-17:
I'm super happy that you folks introduced the ability to load Polyend Tracker Instruments (PTI) yet I would like to see an update which brings us the ability to export PTI – it could be a predesignated folder to make things less fuzzy.  I thinks this would improve the workflow and file organizing. Right now one needs to take note in which project a certain instrument can be found and/or re-organize things via the computer. A simple export PTI would be a super handy shortcut to collect favorite instruments quick and fast and in addition it would save a lot of time because one do not need to navigating thru all the projects one might have.
Thanks

theorize999, 2022-05-27:
+1 for anything to make this process faster, I'm going through a lot of steps right now just to make samples to turn into instruments since I'm sampling the JD-08 digitally in to FT2 clone then exporting out of there and importing into PT, I haven't even begun the process of getting the instruments out of my project files yet. Too many projects not enough time I need to finish my drum sample set that I kind of forgot about for a year, but in that year I've gotten a lot more tools for drums so maybe it was supposed to happen. I guess I should set up a web site for sharing these instruments pretty soon, I'm not good at making web sites though I'm more of a front end guy when it comes to that. I'm thinking about asking Polyend to host them, they seem to be up for that kind of stuff but then I lose the ability to make changes.... if anyone has ideas on distributing options I'd love to hear them, PM me if Github does that I'm not quite sure yet.

Aisjam, 2022-09-23:
+ 1 on this too. If it just bounces the instrument to something similar as the record sample file. its a bit of a process making instruments atm and would be handy to export the favorite presets when you are finished making a track.

danapeks, 2023-01-03:
+1 this would be a vast improvement on current functionality!