polyend/TrackerBetaTesting #1518: [1.6.0b4] LFO clicking

sebpmtl, 2022-05-18:
### Steps to reproduce

On LFO retriggering (by changing Amount in settings page), clicking occurs because of zero-crossing. Also, It'd be great if retriggering/resets  didn't happen by changing the Amount but through the sequencer like in Renoise. 

* How often does this bug happen? (every time, often, "it only happened once")
 everytime 

* Does this bug happen in previous firmware versions?
   yes

* Does this bug happen in every project?
   yes

sebpmtl, 2022-05-18:
yes. You call it recalculate. Is it the same as resetting? the way a normal lfo reset works?

PRAYTHEPO, 2022-05-18:
It's not normal behaviour. We must wait for Polyend to fix it. As i understand, this mess occurs only when you change LFO parameters while song is playing

Jacknametrouble, 2022-05-18:
Does it click on sine?

sebpmtl, 2022-05-19:
> Does it click on sine?

it works well on ''triangle''. It also does it with panning.

patois, 2022-05-28:
> > Does it click on sine?
> 
> it works well on ''triangle''. It also does it with panning. But when using a triangle waveform with panning, it does weird static noises .

There are audible clicks when assigning an LFO to panning I can confirm. I am really getting angry at how Polyend deals with the Tracker. How can they continue adding new features and build new products without fixing the flaws of their current product line in the first place? Audible flaws that aren't addressed by the vendor degrades a music production device to nothing but an overpriced toy. I will definitely sell my unit and it will be the last device by Polyned I will have purchased unless they finally fix all the existing flaws with the 1.6 firmware. It's not like we're at a pre-release version of the firmware at this point...