polyend/TrackerBetaTesting #1521: 1.5 Hanging notes when controlling other synths via midi

77377, 2022-05-19:
### Steps to reproduce

Tried to control a dreadbox Typhon with Tracker. 

When stopping transport on tracker the last played note would just drone on, even if there was a note off event sent after the last trigger. Alternatively after starting or stoppig couple times Typhon would start to drone a note and only stop when transport on tracker was active. 

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
often

* Does this bug happen in previous firmware versions?

no idea

* Does this bug happen in every project?

seems to be project independent.

PRAYTHEPO, 2022-05-19:
Put "OFF" command after last note
@77377

Raadiolained, 2022-06-11:
Many other hardware synths have this issue too, sometimes it can be even synth dependant. 
In your case, it might be that if you press STOP before the OFF message, note hangs on.

It would be good if there were an option in global or project settings to set the STOP press to send OFF messages to all or every active midi channels, maybe even just all the tracks.

closed 2022-06-27 by MortalJS