polyend/TrackerBetaTesting #1522: [Firmware version] Bug title

natct10, 2022-05-19:
## Hi, sorry to bother but 

The fill function is making my screen blink.

### Steps to reproduce

In pattern view, select steps and hit "fill" on instrument and the screen nlinks until I hit "cancel".

Please let me know if you need more detail!

With kind regards

Jacknametrouble, 2022-05-19:
Any chance to take a look at a video showing the issue please?

natct10, 2022-05-19:
Yes sure: https://natct.yourownnet.fr/s/KpaYcrbwys5Agxq

Jacknametrouble, 2022-05-19:
Alright, nice one! Thank you for your help!. I guess 1.6.0b6 is coming soon :D Stay tuned!

natct10, 2022-05-20:
Sure, please let me know if you need further details!

Jacknametrouble, 2022-05-28:
This is fixed in [1.6.0b6]

closed 2022-05-28 by Jacknametrouble

natct10, 2022-05-28:
Thank you, will test the new release soon! Cheers

Jacknametrouble, 2022-05-29:
@natct10 thank you for your help!

natct10, 2022-05-29:
By the way, the new theme looks really nice!

natct10, 2022-05-29:
The issue is fixed, thank you!