polyend/TrackerBetaTesting #1523: Shift + Jogwheel to make changes in higher increments

fulhjort, 2022-05-20:
I think it would be a useful feature to be able to make changes in higher increments if you held shift while using the jogwheel.

For example: if you've loaded a 10 second long sample and want to crop out 9 of those seconds, it's gonna take a lot of jogwheelin', but if you could hold down shift there'd be a lot less.

Wether the incremental change is a set value of 10, 50, 100, or even a global setting, I think is of less concern as long as the function exists.

bambrose, 2022-05-23:
This has been logged before.
Even shift + arrow keys (as an alternative) would be a MASSIVE enhancement at this late stage of the development process.
There are a few areas where scrolling using the jog wheel is unnecessarily laborious :

1. Sample Playback / Granular / Length
2. Sample Edit Screen (where zoom gets you some of the way there - but zooming in and out is tedious)
3. Sample Playback setting a loop

IN certain instances it's just not feasible to zoom out (you have to be close enough to the waveform to work out the correct place.

It's been done on song mode.. and it would be a big improvement if it could also be implemented in the screens mentioned above

trutypesounds, 2022-05-31:
absolutely yes, mentioned this too here somewhere