polyend/TrackerBetaTesting #1525: Step Sequencing on the Grid buttons (like on ELEKTRONs or VOLCAs or everywhere)

fleewortep, 2022-05-20:
I bought Tracker because i've expected this feature to be in it. But unfortunately it wasn't(

I would love to add and change step sequencing using grid buttons while playing like in any other music production device
Press for example PATTERN + NOTE + PLAY (or just NOTE + PLAY) to get into Step Sequence Mode.

[image]
Screen stays the same as in regular mode.

Manually select steps you want this track to play on a fly (1)

Page scroll field buttons (2) always lead-on to be separated from step sequence field visually

If pattern length is 1-32 steps it is fully shown on step sequence field (1)
If pattern length is 33-128 steps use PAGE SELECT BUTTONS (3) to select next pages or scroll wheel

= = =

Please add this feature please

Jacknametrouble, 2022-05-20:
Thank you for your feedback.

fleewortep, 2022-05-20:
> Thank you for your feedback.

sorry but why did you close my submission?

Jacknametrouble, 2022-05-20:
I've closed your submission because it's not going to happen. This is a Tracker, not an Elektron or Korg step sequencer.

fleewortep, 2022-05-20:
> I've closed your submission because it's not going to happen. This is a Tracker, not an Elektron or Korg step sequencer.

oh, i'm sorry man, i didn't want to offend you or anybody by comparing. I just gave a simple to understand example.

If i remove comparing from my submission can you restore it?

Jacknametrouble, 2022-05-20:
This is not about comparison, and no one is offended here, all good, no worries. What I mean is, when you want a plane, you're not getting a car. A Tracker is a tracker, it does what it does in an intended way. The addition as proposed just does not make sense. We appreciate your input a lot, but submitting an idea does not equal it being implemented. Thank you for your understanding. Keep up!

closed 2022-05-22 by Jacknametrouble