polyend/TrackerBetaTesting #1526: [Question] Mono Bass feature

PRAYTHEPO, 2022-05-20:
Not long ago i opened request about Mono Bass feature similar to Ableton's Utility plugin, so here is a question:

Guys, can you please explain why my feature request about "Mono bass" feature was closed as completed, but not implemented in new firmware? If this feature does not correspond with your view of PT, just say so, that will explain everything.

Jacknametrouble, 2022-05-20:
Hey, appreciate your feedback, a lot! You're doing lord's work with many of your tickets. Appreciate a lot! But please understand that we are unable to explain every single case and that sharing an idea is not immediately turning it into a new feature in the next release. In fact, the Tracker is more or less an adult instrument, one that we really enjoy as it is already. Also, its hardware capabilities are already almost maxed out. Then ok, no :) Hope this explains everything. Keep up!

Jacknametrouble, 2022-05-20:
One more thing. As you're so active here, we believe you must be rocking your Tracker hard. We'd love to find out what you're creating with it. And if you'd be interested, you may consider a feature at our polyend.com/whats-up blog where we are trying to create a community around the Tracker itself. If you'd be interested, please drop us a word at jacek@polyend.com Best!

PRAYTHEPO, 2022-05-20:
> One more thing. As you're so active here, we believe you must be rocking your Tracker hard. We'd love to find out what you're creating with it. And if you'd be interested, you may consider a feature at our polyend.com/whats-up blog where we are trying to create a community around the Tracker itself. If you'd be interested, please drop us a word at  ***@***.*** Best!

Email address is incorrect, i cannot send message

With best regards, Dmitry Cherevkov

***@***.***

Jacknametrouble, 2022-05-21:
Hey, it totally is correct and in constant use. This is weird.

PRAYTHEPO, 2022-05-21:
Maybe we can talk here?

With best regards, Dmitry Cherevkov

***@***.***

Jacknametrouble, 2022-05-22:
Hard to say why this is happening. Checked the mailbox and works properly. Must be something at the end of your mailing server.

In regard to your previous question - I've just realized that the Github has changed the way the Close topic function works, not it's by default closing it as realized, while there's now an option to mark it as "not planned". Once again apologies for the confusion caused. Respectfully,

closed 2022-05-22 by Jacknametrouble