polyend/TrackerBetaTesting #1528: Feature Requests...

lerugray, 2022-05-21:
Hi, I was encouraged to leave these here from the PT discord, not sure how feasible these ideas are but figured I may as well ask. 

1. It would be great if we could export tracks by instrument, would make life much easier working in a daw allowing us to make the most of the 8 tracks

2 Some file management options for dealing with samples onboard the machine would be nice

3 I don't know if it's possible, but it would be great one could sample while a track is playing (especially for vocals), or be able to sample a synth being sequenced by the PT.

4 Any way to sample the emulator while its playing, as a bonus it would be great if the emulator was possible to be upgraded to enough of a point that it can run some of the homebrew music ROMS. A Game Boy emulator would also be useful in this regard as there are a ton of music roms for it.

Jacknametrouble, 2022-05-22:
Hey, good to hear from you. Thank you for sharing your ideas. Because of the initial assumptions on the firmware capabilities and what's more important in this case, the hardware capabilities of the device, the points 1,3, and 4 won't be possible to implement on the current hardware. But we'll keep tracking of your requests in mind for the future. Point 2 may be considered. Respectfully,

closed 2022-05-22 by Jacknametrouble

lerugray, 2022-05-22:
Thanks for taking the time to reply! I appreciate it :)