polyend/TrackerBetaTesting #1544: Simple waveform generator in Sample Recorder

bmuessig, 2022-06-01:
It would be absolutely amazing if to the sample recorder, a virtual input could be added that comes in the form of a waveform generator.
Being able to generate sine, triangle and square-waves of different frequencies and volumes would be very useful and should not be too much work to implement. This would allow creating new sounds on the go.

FelixCoffee, 2022-06-01:
That would be awesome!!! :D

bmuessig, 2022-06-19:
After some reconsideration (and rightful advice on Reddit), I don't think anymore that this is the way to go for sample alteration/production. I've bought another synth for that now and will look into getting the tracker working with it then (that makes me wonder if the Tracker can trigger notes on another synth over MIDI while sample recoding).
I'm coming from the software and hardware side with only about 1.5 years worth of piano lessons but great interest in getting to know music production, so I'm still in the right place for silly feature requests :-)

theorize999, 2022-06-23:
yes you can trigger notes on another synth while recording a sample, I haven't had much time for tracking for quite a while but last time I checked it still worked great, this is actually my favorite feature they implemented.... I would have had to sell mine otherwise. This is actually the beauty of tracking.. so yes go for it. It's really fun. You just can't trigger a pattern and sample at the same time you'll have to go for the midi press. If I remember right you can even use an external midi keyboard connected to the tracker and running from the tracker's midi out to your synth, don't take the last one as word I might be mixed up on that one. but the on board keys work just fine for sampling your synth. you can even go one further and create layered patches of your synth by putting equal notes in a pattern of your recorded samples and then resampling that, i reccoment using the note c for all this but do what you want, if you want sample accurate interpolation then at 16 bits 44k resolution sample the note G but honestly C works fine and is way easier to set up. happy tracking.

bmuessig, 2022-06-23:
Thank you @theorize999 for your helpful explanation

closed 2022-07-01 by bmuessig