polyend/TrackerBetaTesting #1547: Chance on FX

clickysteve, 2022-06-02:
### A new feature

Hello.

If you use the Chance effect on a row with another effect, it should apply the chance to that other effect happening.

e.g. Chance and Roll on a particular row. This would mean that the Chance FX would apply to whether or not that roll would happen. At present, there is no way to add probability to an effect. Or is there?

fulhjort, 2022-06-02:
> This is a good idea, but I think that instead of making the chance work on FX instead of notes they should add another chance for FX, so you can have "Chance Note" and "Chance FX"
> 
> Having a chance on notes is very important, so they should add "Chance FX" And not replace the current function

I agree with this.

Jeymadcat, 2022-08-07:
oh yes i want it ! too