polyend/TrackerBetaTesting #1548: Granular Length Step FX

RENEGADE-ANDROiD, 2022-06-02:
I have a lot of experience with the Polyend Tracker and my only request would be for the addition of a Granular Length Step FX.  It could really expand creativity with the granular engine.  Being able to adjust a length value per step would allow for some very interesting and cool sound manipulations.  I'd appreciate the consideration of this suggestion.

NoSitRecords, 2022-06-02:
This!!

bambrose, 2022-06-02:
Yeah would be amazing if possible

trutypesounds, 2022-06-02:
YES

CarloAvanti, 2022-06-09:
Really missing this feature !!

Undistraction, 2022-12-21:
Came here to ask for the same thing. With the right settings, manually manipulating the length gives lovely results, especially moving around very low values. It would be wonderful to allow Granular length to be manipulated by FX and by LFO/envelope for that instrument.

solipsvs, 2024-01-05:
yes please!  give it a cc too!  make it a target of modulation and fx