polyend/TrackerBetaTesting #1551: [1.5 and in beta 1.6.0] random instrument fx not working properly

b00gieboarder, 2022-06-06:
i've only had this brilliant machine a week and I'm completely obsessed...but seem to have found a bug.  Bear with me if this has been reported before (or if i'm misunderstanding the function of this fx) but I didn't see it in a search.  Bug is reproducible in both 1.5 and beta 1.6.

### Steps to reproduce

i Random instrument FX is not properly constraining the range of instruments.  Whether in constant, From-To or random the generated randomization of instrument will not stick to the desired range that it has been told to stay in.  

For example, I want only the hi hats on instruments 3-5 to be the option that the fx track chooses to play.  But without fail the fx is including instruments from any instrument track.  Even if I tell it to randomize the instrument with just Constant Instrument 3, it will include the entire range of instruments in it's choice as to what plays.   

Bug appears every time in both firmwares 1.5 and beta 1.6 and in the two projects I have attempted to reproduce it in.

strangnet, 2022-06-09:
I believe it to be relative to the instrument chosen on each beat.

To alternate randomly between three different kick samples in sample slots 7-9, I achieved it with:

`C5 08 i1`

The `i1` means choose randomly +/- 1 relative to sample 8, i.e. 7, 8, and 9.

b00gieboarder, 2022-06-09:
Very helpful thank you Patrick!  I'll give it a try!  

🔥

closed 2022-06-11 by Jacknametrouble