polyend/TrackerBetaTesting #1557: Keep track of your projects until you reach 100

dias236, 2022-06-13:
## Hi, I'd like to suggest a feature for the projects screen!

I didn't check if this has been asked before, but in facebook the issue made some people aware of this issue/limitation. Most of them , like me, didn´t have a clue of what was going on after saving more than 100 projects and realizing they disapeared from sight in the PT. But, you could see them in the SD Card on a computer.

### It would be very intuitive to have a count of the projects we have on the sd card in the projects screen.

A text, in this screem, advertising the limitation of the display of only 100 projects would save some time and paranoia. You could also use this text to encourage doing back up of old projects to manage available space.

Dispalying a current count would be great. Example: "**Displaying 50 of 100 projects slots**".
And, in case one user is saving too much projects like f.e. "**Displaying 120 proyects of 100 projects slots**" you guys could make another sentence appear like: "**Start making back ups now**!"

I think this is a simple thing to do that will not take lots of limited resources the PT has. It would also make it more user friendly in the project management aspect of thing. It would definately avoid confusion for those, like me you don't read the manual hahahaha!

Maybe **if the list of project were to be numbered** it could also be more intuitive. But maybe I'm asking too much here.

Well, it's all for now. I hope this feature request finds you in great moment of motivation and creativity!
PEACE

tfspsound, 2022-06-13:
Would it be possible to handle projects in the same way recordings are handled - by just making a projects2, projects3 etc?

mihalisan, 2022-06-27:
In general, the project management suffers and its a shame for such a machine.  The 100 project limitation doesn't make much sense given you can have 400 GB  sd cards.  Why not 100 projects per folder lets say, which would make things much easier.  Any improvement in this respect will be much appreciated, as project management gets really tedious.

arphex, 2022-06-27:
In general, the project management suffers and its a shame for such a machine.  The 100 project limitation doesn't make much sense given you can have 400 GB  sd cards.  Why not 100 projects per folder lets say, which would make things much easier.  Any improvement in this respect will be much appreciated, as project management gets really tedious.