polyend/TrackerBetaTesting #1559: [1.5.0] Clicking sound when changing volume

notaprogrammer9, 2022-06-18:
## Hi, sorry you had trouble!

### Steps to reproduce

When two sounds are sequenced after each other, the beginning of the second sound results in a click when its volume is lowered. Volume change in Instrument Parameter and volume change in the sequencer, both result in a click. 

I assume the volume change kicks in too late. 

Click disappears when a volume change is programmed right before the sound or when the sound is copied to an empty track. 

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

Every time

* Does this bug happen in previous firmware versions?

Not in 1.3.0

* Does this bug happen in every project?

Yes.

sebpmtl, 2022-06-18:
I've experienced this before, but last time I used my tracker with the new firmware, I was happy that finally, the lfo didn't click anymore. But I remember, when you automate the volume, or lfo cutoff, from a lower value to a higher value and vice versa, you get clicks. As you said, clicking all over.

voidshine, 2022-07-14:
This may be a general issue with volume changes of any kind, including panning changes (pan affects left and right channel volumes). I tried to do a volume fade with a sequence of volume FX commands. click click click click click as it stair-stepped volume instead of smoothly fading as expected. Applying a triangle-wave LFO to panning on a wavetable instrument, I could hear frequent clicks as the pan changed.

trutypesounds, 2022-07-15:
yeah clicks still in latest fws, thats sad =/

closed 2023-04-27 by MiroPoly