polyend/TrackerBetaTesting #1560: [1.5.0] Monitor input volume too low or too high and distorted recording

notaprogrammer9, 2022-06-18:
## Hi, sorry you had trouble!

### Steps to reproduce

The input monitor volume resets when exiting the sample recording screen. The input is close to inaudable when the tracker is playing back internal audio. 

Adding gain on the external device (connected to the input) results in clipping. 

Recording at a low volume results in aliasing and high pitched whine in the recording after normalizing the recording. 

When the bass in the Tracker is too high, the Tracker sounds distorted. The headroom needs a rewrite. 

* How often does this bug happen? (every time, often, "it only happened once")

All the time

* Does this bug happen in previous firmware versions?

Yes

* Does this bug happen in every project?

Yes.

trutypesounds, 2022-07-01:
mentioned a long time ago here #1172 and there

theodorozzz, 2023-01-12:
Hello! I have the same issue, but in one of projects it works as it should!
I mean line input signal plays in pattern mode without issues in of projects. Have no idea what the difference between them

theodorozzz, 2023-01-13:
Ha. I found the reason. I miss that it have also fader in master section for line input. My bad! Solved for me though)

closed 2023-04-27 by MiroPoly