polyend/TrackerBetaTesting #1572: Kit Mode

jsensurreal, 2022-07-02:
It would be nice to be able to make custom kits where you can assign each pad to play a specific note/instrument. Groups of samples could then be played at the same time and recorded instead of recording one sample at a time. This could also be useful for playing multiple samples live at the same time, like a drum kit, or multiple slices of different samples together.

Jacknametrouble, 2022-07-02:
You can already do this with the Polyend Tracker using Slice Mode with a prepared wave file.
https://www.youtube.com/watch?v=bn4vPrTU9js&ab_channel=Magnetosphere
https://polyend.com/whats-up/tracker-beat-slice-midi-tutorial/?pages=3&catid=101

closed 2022-07-02 by Jacknametrouble

jsensurreal, 2022-07-02:
I am talking about a kit using multiple individual samples. So you can still control instrument parameters but playback multiple samples at the same time with the pads. It is not the same thing.

Jacknametrouble, 2022-07-03:
We are always keen on hearing user feedback about possible functionality improvements, however, we are not obligated to implement such feature requests. Please respect that while we appreciate our customers’ opinions and wishes, we cannot guarantee or promise to implement them.

Polyend Tracker was made with some initial assumptions in mind. Its hardware resources are limited, thus not every “simple” (from your perspective as a user) feature request is really simple, or sometimes it is not even possible to implement. Thank you for your understanding.