polyend/TrackerBetaTesting #1574: [1.6 and prior] Reverb gets incredibly distorted at high ‘times’ and low ‘diffusion’

ryancara, 2022-07-06:
I've just noticed that when the reverb time is long and diffusion is at a low setting, I get lots of crackling and distortion at the
output that can't be stopped until I press play on the pattern.

I’ve only just noticed it while using the Tracker as a synth, but reverting back to 1.5 produces the same problem.

Cheers,

Ryan

ryancara, 2022-07-06:
Here's an audio file demonstrating the issue:

https://drive.google.com/file/d/1PrJPsQuWIVX2m9EV1CN_9KQi2IMTcT-4/view?usp=sharing

PRAYTHEPO, 2022-07-06:
Can assume that this is normal behavior if diffusion is set to very low values, cuz it becomes very fast unfiltered delay. Any type of reverb is smudged delay, dont forget about it

Jacknametrouble, 2022-07-06:
Thank you for your comment @PRAYTHEPO

closed 2022-07-06 by Jacknametrouble

ryancara, 2022-07-07:
Before it goes into complete chaos it starts clicking and popping. It makes me think that it's a CPU issue.  I'll upload another example later today!