polyend/TrackerBetaTesting #1575: USB audio interface

bmuessig, 2022-07-06:
It would be quite useful to be able to have a USB class compliant audio interface on the tracker to sample from and to duplicate the line output over. This would allow easier interop with computer software without requiring an external audio interface and without loosing audio quality in either direction.

USB class compliant audio is available in the standard Teensy audio library and should therefore be very straight-forward to implement and enable. This would help de-clutter our desks and enable to more easily (and losslessly) record the tracker's output.

bmuessig, 2022-07-06:
I fully understand why the Tracker's firmware is not open-source, but this would be one of these things that I'd totally take care of myself if it was :-)

ckamp1, 2022-07-06:
This would be awesome, no class compliant USB Audio is one of the biggest things I miss from the tracker