polyend/TrackerBetaTesting #1581: Track Mixer window improvement

PRAYTHEPO, 2022-07-12:
That would be nice to use combination Shift + Jogwheel in Track Mixer window for decreasing/increasing volume off ALL tracks simultaneously

tfspsound, 2022-07-13:
How about the addition of a latching option so we could adjust groups of tracks?

tfspsound, 2022-07-14:
> > How about the addition of a latching option so we could adjust groups of tracks?
> 
> @tfspsound this already exists. Push few buttons dedicated to tracks simultaneously and scroll the jog-wheel. But it is hard to push all eight buttons to lower volume of all tracks

Do you mean push and hold while turning the jog wheel? 
If so that isn’t what I am suggesting. Highlighting just tracks 1 and 8 or 1,7 and 8 is equally as difficult as doing all 8. With a latching option we could create groups to turn tracks down together that might be difficult because of the way the samples have been added (alternatively a way to rearrange the tracks would be useful) 
By latching I mean pressing Shift+Screen buttons eg Shift + Tracks 1,3,4,7,8 (screen buttons could be added one after another not simultaneously) then turn the jog wheel to adjust the level, then holding Shift again and pressing the screen buttons to release them from the group.

PRAYTHEPO, 2022-07-15:
@tfspsound understood. Unfortunately, Shift + Track is already Solo Track option. So we can only beg Polyend for all-in-one option

tfspsound, 2022-07-15:
@PRAYTHEPO your right I’d forgotten about solo I’ve been without my tracker for a few week while it was being repaired. An option within config would be good in that case. Like you said an all in one would be great.

closed 2022-07-21 by Jacknametrouble