polyend/TrackerBetaTesting #1585: [1.6] adjacent steps low-velocity popping

thumperjack, 2022-07-20:
if you take any sample and on every other step reduce the velocity down below ~25, when the steps with reduced velocity play there is an ever-present pop at the start, you can hear the pop more clearly if you put the offset steps velocity below 20. Even if you slow the tempo down to 10bpm, so the full velocity sample is long gone, when the lower velocity step plays it sounds like the pop from a non-zero crossing, it shouldn't pop like that, it sounds like something is sticking, like an amp envelope is staying open trying to carry over

PRAYTHEPO, 2022-07-21:
https://github.com/polyend/TrackerBetaTesting/issues/1559

Issue already exists with multiple variations. This is the one of many ⬆️

MortalJS, 2022-07-25:
duplicate #1559

closed 2022-07-25 by MortalJS