polyend/TrackerBetaTesting #1591: [Firmware v1.6 and previous] Please sort projects by name / Random projects sorting

natct10, 2022-08-19:
## Hi, sorry you to bother you with this, but it really makes the tracker unpleasant to use.

It would be *really* nice to have the projects sorted by name instead of being randomly spread in the folder. It would be *really* nice to have the same behaviour we get when we browse samples or recordings.

When you have 80+ projects, it's becoming really painful to look for a specific project because there no -or I didn't get it?- sorting there.

### Steps to reproduce

Go to `File` / `Open` and try to find a specific project.

### Other important information

It would be awesome to sort by last modification, but alphabetic order would be really nice!

Thanks in advance and please let me know if you want to discuss this further.

Have a good day!

jdlesemajeste, 2022-09-15:
Hello, I would also like to see this implemented, I would really enjoy having this feature.

briandgregory, 2022-09-15:
Yes, the ordering of projects seem "sort of" like most recently created, but not always. Ideally sort or search options would be really nice.

TheGrizard, 2022-09-26:
I would be keen for these sort options and also to see extra metadata visible on the file browser screen including date created, date last opened/modified and a tags/comment section so we could annotate tracks with an artist or set or album name etc to enable grouping/id of projects

PRAYTHEPO, 2022-09-26:
@TheGrizard 
There is no date/time counter in CPU of PT, so it is not possible. Polyend commented few times about it

danapeks, 2023-01-03:
Alphabetical sorting been a request since 2020 now... Come on Polyend please.

natct10, 2023-01-03:
> @TheGrizard 
> There is no date/time counter in CPU of PT, so it is not possible. Polyend commented few times about it

Thank you for your reply! Sorting projects by name should be possible then?