polyend/TrackerBetaTesting #1592: slide to note with speed xx

impulsetracker, 2022-08-20:
hello , dont know if polyend tracker has this feature ?  thanks .
in schism tracker is this effect > (GF0) 
please see the video i recorded because it will be more clear than my words:

[https://youtu.be/vaMF-8Jq9U0](url)

thanks

impulsetracker, 2022-09-08:
> 

thank you for your clarification, I didn't know how to express it, even in my language hehe. Indeed, that's the behavior I'm looking for so that the samples (with filter envelope included) can sound organically and fluidly through the song. I think I've tried the parameters you mentioned but I haven't been successful.  The only way I have found is to do it "by hand" using the "P" (position) command and calculate "by ear" how I would like the sample to sound through the pattern. Thank you very much again. 

Translated with www.DeepL.com/Translator (free version)

PRAYTHEPO, 2022-09-08:
You are doing extremely hard work with this :)
Watch my video, I hope you would understand how to use Slide Up FX. I think this is what you are looking for.
Also, you need to keep in mind that every 16 is a half-tone and calculate a little bit in your head

https://user-images.githubusercontent.com/58429366/189165504-a720b685-0ed7-4969-890c-24c1addcf3da.MOV

closed 2022-09-08 by impulsetracker

impulsetracker, 2022-09-08:
https://youtu.be/9x0JOA2vk6Q
[https://youtu.be/9x0JOA2vk6Q](url)

PRAYTHEPO, 2022-09-08:
In my video sample does not retrigger, thats why i sent it to you. To show you how to achieve your desired results
Using GLIDE FX sample retriggers, using SLIDE UP/ DOWN DO NOT RETRIGGER.
In instrument playback parameters i chose Ping-Pong Looping.

impulsetracker, 2022-09-08:
thanks bro. let me show you what i have done with the slide up fx you said me: 

https://www.youtube.com/shorts/3yICFdqHDzo

that is a way to do what I want , but it is not dynamic as I showed you  but very "artificial" and "manual" , you understand me right? , because you have to change the glide value at every moment to fit the note I want , and the latter I can only get it "using my ear" hehe . Sorry if I didn't understand you. sorry for the inconvenience. Send me that video of yours if you want, it will be a pleasure to see it.

PRAYTHEPO, 2022-09-08:
Yeah, i understand. Its like glide instrument option in Renoise, so that notes are always transpose smoothly to each other even when yiu play another note. Sadly, this is not possible on PT. Slide Up/Down FX is the only way to mimic something like this. I think that Polyend will not implement this "gliding" feature, cuz memory limit for further firmware updates is almost full. 
(They can delete useless features as like NES emulator, but they wont. Its "too much work")

impulsetracker, 2022-09-08:
your way of solving it is correct, I had already thought about using slide up and slide down, the problem is that the fast music (fast tempo) that I make requires fast note changes and to do that with slide up and slide down would be hell as I showed you in the previous video hehe. If it was a slow tempo music , then yes, then I could use that effect you told me and it would be less hellish hehe.  Thanks anyway. It's funny that such a "simple" program like schism tracker or impulse tracker seems to have more power and more versatility than this polyend hardware. Thank you very much