polyend/TrackerBetaTesting #1594: [1.6.0] difference between audio output and audio exported

tomasoma, 2022-08-24:
## I notice a significant difference between the rendering of the audio output and the result of a song export

I link a dropbox folder with 2 files of the same song, in the same state of the project, all exactly similar. one is the song export "master.wav" and the other is the output recorded and normalised (-0.3 db) "PT24B1.wav" 

https://www.dropbox.com/sh/s8h3s13rfdxm9zd/AAAfg21hAvPNyiYa-11tM6z_a?dl=0

PRAYTHEPO, 2022-09-02:
@tomasoma what to look at? What you want us to notice? Difference between volume level?

tomasoma, 2022-09-03:
@PRAYTHEPO No, Please ...
I added 2 files in the same folder : 4 bars cuted from each files at around 0'54" 
take some correct headphones, i have cheap JBL and i can't miss the difference : 
Export : more volume, the fx are far away
Output : less volume, the fx are louder.

Even if the volume level is globally louder on the export, why is the reverb so far away ?
I guess the difference of global volume is pretty normal, but shouldn't we able to trust the output to mix the fx to export the song ?
clearly the fx are not treated the same way.

I hope my comment is clearer now.

PRAYTHEPO, 2022-09-03:
@tomasoma i will check a bit later. I noticed some difference at reverb level, but was listening through JBL in-ear headphones. Will check through my AKG. Interesting...

PRAYTHEPO, 2022-09-07:
@tomasoma
Hi again. Listened carefully through headphones. Yes, difference is obvious, but i think this is not a bug. I think that when you export song it does some audio processing which we don't know about

tomasoma, 2022-09-07:
Hi @PRAYTHEPO
well really ? 
So I repeat my question : shouldn't we able to trust the output to mix the fx to export the song ?

> Yes, difference is obvious, but i think this is not a bug
I don't know the exact definition of a bug, but this is still a problem as a user, because i expect to be able to trust the output to make my mix and export my song.

> when you export song it does some audio processing which we don't know about
Aren't you a Polyend staff ?

PRAYTHEPO, 2022-09-07:
@tomasoma no, i am just community member. Are you sure that you didn't tweak anything right before export? Cuz i just made a comparison of my own mix through output recording and export, and they sound pretty much the same, as i can hear. Which firmware version are you using?

tomasoma, 2022-09-08:
@PRAYTHEPO ok
Yes i'm totally sure i didn't tweak anything between export and output record.
the version of the firmware is in the title : [1.6.0]
Polyend staff seems to have deserted this github trackerBetaTesting pages :/

PRAYTHEPO, 2022-09-08:
> @PRAYTHEPO ok

So it's very strange. Yesterday i exported my new sketch and i don't see difference between usual output and exported .wav.
Which revision of PT u use? Maybe this is the case. (As i know, the hardware is the same, only the case got changes from metal to plastic. But who knows...)
My main conclusion that this is some audio processing on finalizing song
> Yes i'm totally sure i didn't tweak anything between export and output record.

Sry, didn't see 😅
> the version of the firmware is in the title : [1.6.0]

Yeah, dust covers this pages for a month after they released Polyend Play
> Polyend staff seems to have deserted this github trackerBetaTesting pages :/

MiroPoly, 2023-04-13:
This is by design: export happens before Bass Boost, Space and EQ master FX.

closed 2023-04-13 by MiroPoly