polyend/TrackerBetaTesting #1596: [1.6.0] Graphical Stuttering/Freezing of Polyend Tracker

jcnrn, 2022-08-25:
Following clickysteve's issue of sporadic freezing, I also experience a sporadic graphical stutter in operation of the Polyend Tracker on version 1.6.0. I have attempted to power via the Polyend cable, a USB A to C cable using an iPhone wall wart, a Motorola USB C fast charging cable, and a different USB A to C cable via my desktop computer and was unable to resolve the freezing. As can be seen in the video below, I have graphical stuttering while the audio remains fluid. This occurs not only during song playback but through general operation of the Tracker where it interrupts my control of the unit.

Video of Stuttering (please excuse the chiptune lol)
https://drive.google.com/file/d/1TqtPPpjB1-zgeGW1YGFa4AfS156NDru2/view?usp=sharing
Stuttering occurs at 1m23s and 3m08s

_Originally posted by @jcnrn in https://github.com/polyend/TrackerBetaTesting/issues/1259#issuecomment-1226425998_

PRAYTHEPO, 2022-09-08:
Interesting, Never happened to me before. Is there any other projects in which you experience this freezes? 
If it is only in this project, I assume that maybe:
- your in-project memory is almost full
- slow MicroSD card
- bad power
- some hardware related glitches

I am not Polyend staff, but the case is interesting, cuz it seems like PT works fine except of graphical freezes.
Did you try to reflash your firmware?

jcnrn, 2022-09-08:
Thanks for your interest and taking the time to comment: 

- I'm currently using a two-month old 128GB SAMSUNG EVO Select Micro SD-Memory-Card (microSDXC 130MB/s). and it is only filled with around 20GB of data. I purchased this attempting to resolve this issue. 128 is [cited](https://help.polyend.com/784248-What-capacity-SD-cards-can-be-used-and-whats-the-file-partition-format-used) as the maximum tested card for the PT on Polyend's website
- The project I'm demonstrating on uses 3 single cycle waveforms and 5 drum samples formatted for an Amiga tracker. I believe the memory load was around 3%. 
- I've run the PT direct from outlet at two houses, used a few power strips (at one house), and run directly from my desktop computer which uses a substantial EVGA power supply.
- This is actually my second PT. I RMAed my first one (V2 I believe) because of the jog wheel issue and received a V3. My original one did not have this issue, but the new one does. I believe my RMA happened around version 1.4.

I'm in the middle of a move right now, but will reflash the firmware as you recommend and try the original SD card and some other smaller-sized ones to see if that helps. Will follow up.

PRAYTHEPO, 2022-09-08:
@jcnrn waiting for your response. Really strange behavior, cuz freezes were in 1.6.0b3-4 and now they are fixed. But your type of freezes is new to me

MF303, 2022-09-29:
I have the same problem with the 1.6 firmware.
The screen start to freeze up sporadically after running a pattern for quite some time. (not even touching any buttons)
And over time it gets worse and if you let it run it eventually stops scrolling completely (audio keeps running fine) and also all buttons getting unresponsive or stop working completely. Only way out is reboot the device. 
With 1.5 firmware I could not reproduce this problem.

I tested with quite a lot of my projects and a few that came with the device from very simple to very complex ones.