polyend/TrackerBetaTesting #1600: External audio (line in) glitch when you press play

teamoth, 2022-09-07:
Hi there. If I'm feeding audio into the Tracker from another device, when I press play on the Tracker the audio glitches momentarily. 

This happens even when there's no instruments loaded, the sequencer is entirely empty and the external audio is not going through the reverb or delay.

Apologies if this is already a known issue. I couldn't find it by searching here.

I'm using v1.6.0b4.

PRAYTHEPO, 2022-09-07:
@teamoth Did you try to replicate this bug on 1.6.0 official release?

teamoth, 2022-09-13:
@PRAYTHEPO Sorry it's taken me a while to get back to you. Yes, it does this on 1.6.0 also. I've just tested it.
If you have any audio playing through the line-in, when you press play on the Tracker the audio coming through the line-in momentarily drops/glitches.

teamoth, 2022-09-13:
Would be great if some other users can try this out to see if they have the same issue.
My Tracker is one of the very first ones.

PRAYTHEPO, 2022-09-13:
> Would be great if some other users can try this out to see if they have the same issue.
> 
> My Tracker is one of the very first ones.
@teamoth 
Will write you back once i check this glitch/bug