polyend/TrackerBetaTesting #1601: [1.6.0] Serious bug in the Sample Recorder

bmuessig, 2022-09-14:
### Steps to reproduce

In the Sample Recorder, press and hold the Note button. The background color will change and get darker.
Then press and hold the Instrument button and begin to scroll through the MIDI channels.
After a few iterations, the Tracker will crash completely and only a power cycle will (most of the time) bring it back to life.

**Power cycling will not always bring it back, though.**

I've seen the screen get out of sync and display changing patterns and stay this way on many subsequent reboots **even after multiple power cycles**:

[image]

### Other important information

* How often does this bug happen?
Always after a few cycles of the knob while holding the Instrument button.

* Does this bug happen in previous firmware versions?
I didn't observe it there. Maybe it was already present.

* Does this bug happen in every project?
Yes.

bmuessig, 2022-09-14:
@PRAYTHEPO Thanks for the quick response. This image was taken after about two power cycles (plugging USB-C out and in again).

PRAYTHEPO, 2022-09-15:
@bmuessig 

I am too scary to reproduce it by myself to see how it behave 😅
We need Polyend to recall that they have PT GitHub. It really looks like it can damage your unit, especially when you said that artifacts do not go away immediately after reboot. It's something on hardware side or it is some memory glitches

idchlife, 2022-10-17:
You can record sample to different notes/instruments by holding pads? I never knew that... Not trying it now, though. Bug seems very serious

bmuessig, 2022-10-17:
Polyend does not really seem to care about this. I tried to contact them via their support system to inform them of this rather serious issue, but they have only responded with "use the original power supply", which I always did.

idchlife, 2022-10-18:
@bmuessig well it's not like this is an opensource project... The Polyend Tracker as of right now is a complete product with existing bugs that will be fixed in 1-2 years and it is ok for a product like this.
I mean I'm trying to push forward with tracker making weird stuff with effects and sounds and still it did not fail on me. And I actually expected it to fail because well, it is a software on a low-end hardware :D

I even have a bug when my songs simply do not export. Device freezes forever. But I just record sound externally. This is not a major bug IMHO because if one has ability to insert sdcard to get the .wav file from tracker - one have 3.5 jack socket or dongle to record external sound.

bmuessig, 2022-10-18:
I've been in contact with Polyend again and they have passed it on to the developers.

> The Polyend Tracker as of right now is a complete product with existing bugs that will be fixed in 1-2 years and it is ok for a product like this.

It certainly is, but for issues that crash the device in such a way, there should be fast tracked resolution.

closed 2023-04-21 by MiroPoly

bmuessig, 2023-04-21:
@MiroPoly I'd appreciate any input on what caused the issue and whether it caused permanent damage to my unit, which I am currently having an RMA on.
The issue re-appeared multiple times and only firmware re-installs brought it back each time.

MiroPoly, 2023-04-21:
@bmuessig Thanks for your message. We are sorry you experienced this issue. As stated here, we logged the issue and will do our best to deliver a fix in the upcoming update(s).

I also just checked with our customer support agents and I believe they told you the same. Forgive me if I missed something.

After reproducing the issue ourselves, we think it cannot cause permanent damage. Possibly your unit is damaged for a different reason? If that's the case, please get in touch with our customer support again and they will be able to help you.

Thanks for your patience and support in this matter.

FYI https://github.com/polyend/TrackerBetaTesting/issues/1635