polyend/TrackerBetaTesting #1604: Allow instruments to have a MIDI in channel assigned to them in Instrument Synthesizer

SeanHub, 2022-10-11:
First of all, I love the instrument synthesizer feature that was added in 1.6.

It would be a game changer if it supported the ability to assign up to some of the instruments to a MIDI in channel. This would allow the tracker to be a sample/wavetable multitimbral monster.

For example, pressing shift + instrument parameters takes me into instrument synthesizer. All my instruments are listed on the right. Since the tracker is limited to 8 notes being played at the same time, we can in theory assign up to 8 instruments with a MIDI in channel. After assigning MIDI in channels, the instrument synthesizer allows me to play my various samples on different channels. This would be amazing for live performances controlled by other gear (Keystep Pro or Ableton or whatever other sequencer/performance tool you use). In my scenario, I use a combination of VSTs, Hydrasynth and Polyend Tracker. Using my sequencer I could assign MIDI channel 1 to the Hydrasynth, channel 2 to a VST on my computer and channel 3, 4, 5 and so on to the tracker for all the other instruments.

I would love to use the tracker as a sampling beast in my studio, with this feature, the tracker would be amazing. Please consider this.

Monsieur36, 2022-10-21:
I would absolutely love to see this.

Monsieur36, 2022-10-28:
Alongside specific instruments being assignable to MIDI channels, it would be very useful and interesting to allow each track to have its own MIDI channel so that incoming MIDI on the respective channel would essentially play whatever instrument was programmed on the track. :)

sabjorn, 2022-11-28:
This would be ideal. The current implementation does not play very well with my other machines. I really would expect each midi channel to have a separate “voice”.

adriananders, 2023-02-07:
+1. Specifically I would like assignable multi-timbrality in the MIDI-synth mode, making the PET closer to a Digitakt when integrating it with a DAW.