polyend/TrackerBetaTesting #1614: Solo available in pattern mode..

pasildon, 2022-11-17:
This would be a quick workflow feature so you can stay in pattern mode to solo and mute without having to switch pages and exit pattern mode to solo!

Synth-Maarten, 2022-12-12:
In pattern mode “Shift+Play” has no action tied to it, its the same as just pressing “Play”.
So maybe “Shift+Play” could be used to bring up SOLO for the screen keys in pattern mode?

danapeks, 2023-01-03:
> In pattern mode “Shift+Play” has no action tied to it, its the same as just pressing “Play”. So maybe “Shift+Play” could be used to bring up SOLO for the screen keys in pattern mode?
That's not actually correct as shift + play plays FROM the row you are on. Useful feature
Agree with the concept though, would be great to have Solo in pattern mode.