polyend/TrackerBetaTesting #1615: Copy /paste/duplicate “blocks”in song mode!

pasildon, 2022-11-19:
This would be a rapid workflow feature if you could copy/paste within song mode.  With the song blocks  creating a NEW pattern for each block copied and pasted. Then you could easily shift and rearrange and even remix rapidly by shifting blocks around all within one song screen.

PRAYTHEPO, 2022-11-19:
If i understand correctly, you want to copy/paste blocks from existing pattern into a new pattern. You can do that. Just select needed block with arrows and press copy. If you need several blocks - hold down shift and select with arrows

pasildon, 2022-11-19:
> If i understand correctly, you want to copy/paste blocks from existing pattern into a new pattern. You can do that. Just select needed block with arrows and press copy. If you need several blocks - hold down shift and select with arrows

Thank you yes you are correct I’ve been trying this and it somehow wouldn’t work. But it indeed does and does exactly that, creating a new pattern thanks 😊

closed 2022-11-19 by pasildon