polyend/TrackerBetaTesting #1618: Alternate Filter Models/Algorithms

dustboxednorth, 2022-12-06:
for the past year, I've been using the tracker for almost all of my musical ideas. however, one of my pet peeves is the filter. in my opinion it's very lacking and it could be really nice to have more options available.

as someone who enjoys sound design, having different filter models available to me would be a godsend. sometimes I want to put a filter envelope on a single-cycle waveform to make a bass or something and I just can't get it to sound quite right because the filter is basically "just a filter;" it does it's job, and that's fine, but it doesn't give you much room to nitpick.

maybe I'm just coming at it with the wrong perspective though.

PRAYTHEPO, 2022-12-06:
As any thing that Polyend develops: it is just working, it is enough.

I would be really impressed if they will develop 1.7 firmware with some enhancements. As one of the devs said: there is no room for improvement, storage space for firmware is limited.

Forget about future updates. You bought golden dung, they got your money. Nothing personal, just business

dustboxednorth, 2022-12-09:
> As one of the devs said: there is no room for improvement, storage space for firmware is limited.

source? genuinely curious where this came from.
> You bought golden dung, they got your money. Nothing personal, just business

well okay then!

dustboxednorth, 2022-12-09:
> @dustboxednorth 
> Sorry, but source is somewhere in discord of Polyend Tracker
> You can ask people in it, they will say the same as me

I see. this is news to me btw; I didn't even know there _was_ a Discord server.