polyend/TrackerBetaTesting #1619: [noticed since 1.4] LFO fx (g/h/j/k/l) value not corresponding with Instruments Parameter LFO steps value

CarloAvanti, 2022-12-06:
Hi,

I noticed this thing since a while now, never had the time to make a video... I don' t know if it' s really an issue or is something I don't understand.

Basically the thing is that value of the steps of LFO in instruments parameters is not the same as the LFO fx in the fx column.
So for instance, If I select 16 steps in the Instruments Parameters, to have the same effect I have to use the value g24 in the pattern to have the same movement of the LFO. The same happens with all the values, and for all the other LFOs

Its still usable, because you can really hear the difference and if you know you can go by logic, but still very confusing.

Here a video where you can understand what I mean: https://www.dropbox.com/s/tj9t27v5myuc36e/IMG_1106.mov?dl=0

sebpmtl, 2022-12-12:
Trackers work with ticks.so 24 is right.

CarloAvanti, 2022-12-14:
Ok, thanks, I didnt knew it, very interesting.
Still I dont understand the logic of using ticks for the FX and steps in the instruments parameters. :) But at least now it make sense.

closed 2023-04-26 by MiroPoly