polyend/TrackerBetaTesting #1622: Firmware development ongoing?

SandyMental, 2023-01-05:
Or has it been abandoned by Polyend?

Would be kind to get some words from the devs/Polyend on what to expect or not anymore.

PRAYTHEPO, 2023-01-06:
> Or has it been abandoned by Polyend?
> 
> 
> 
> Would be kind to get some words from the devs/Polyend on what to expect or not anymore.
> 
> 
> 
> 

Last update and contact with Polyend was in July 2022. Answer is obvious

SandyMental, 2023-01-06:
> > Or has it been abandoned by Polyend?
> > Would be kind to get some words from the devs/Polyend on what to expect or not anymore.
> 
> Last update and contact with Polyend was in July 2022. Answer is obvious

yeah obvious to people lurking around here, but would bet that's not the majority of their  customers ;-)

zikajk, 2023-01-12:
I had the same question.
I was hoping that the development for Play would also be reflected in Tracker. 
The software seems to be quite similar, so it would be possible to fix the nasty bugs that the Tracker software has. Unfortunately, it seems that the products are not very related and the Tracker will not receive an update for a while :-(. But it is not dead according to Polyend, which is good :-)
Below you can see a reply from Polyend team.

_Thanks for reaching out. I'm sorry to hear you are not satisfied with our product even after the 1.6 firmware update. Tracker development is a bit slower right now since we are concentrating on Play and its growth and as we are a small team we can work on a limited number of projects. There will be a firmware update in the future at some point, but as for now, I can't tell you when exactly. Thank you for understanding._

synthexpert, 2023-02-01:
> Thanks for reaching out. I'm sorry to hear you are not satisfied with our product even after the 1.6 firmware update. Tracker development is a bit slower right now since we are concentrating on Play [...]

That's reasonable after all. It would just be fine to close all the most annoying bugs...

swindus, 2023-02-19:
Don't expect anything. I received a similar statement form Polyend regarding development of Medusa firmware and now it's legacy.