polyend/TrackerBetaTesting #1630: SDK for future firmware enhancement

maks, 2023-01-20:
In [this comment that relayed a message](https://github.com/polyend/TrackerBetaTesting/issues/1622#issuecomment-1379950858) from @wjakobczyk-polyend staff it was mentioned that Polyend is a small team of developers who need to focus their resources on there most recent product which is completely fair enough.

But given the ability to flash alternative firmware on to the Tracker, it would be amazing if Polyend could release if not the full source code for the firmware, a SDK of some sort that would allow eager enthusiasts in Tracker community to continue to work on improving the Tracker firmware. A great example of what can be possible is [how Novation did this for the firmware of their Launchpad pro](https://github.com/dvhdr/launchpad-pro).

It of course could be made very clear that people do this completely at their own risk and its not at all supported by Polyend, but it would at least give the opportunity to keep enhancing this wonderful piece of gear.

SandyMental, 2023-01-20:
the source was leaked some  10?months ago - just because PE did not secure their website properly - everbody could download it all the time, technically seen.

maybe youl'll find the files somewhere on the web, but actually PE did not approve in anyway, so my guess is:
that wont do that at anytime, as they seem to not care about user feedback too much.

see no one of their Devs seems to look into that repo anymore.

PRAYTHEPO, 2023-01-20:
> the source was leaked some  10?months ago - just because PE did not secure their website properly - everbody could download it all the time, technically seen.
>
>
>
> maybe youl'll find the files [...]

Yes, source code exists in web, but it is too old, maybe 1.0 version.

Finally someone see that Polyend cares about money, not idea and customers. They reinvented their first revision of Tracker with metal body and square caps, and called it Silver Edition for double price. Pathetic. Gold diggers

zikajk, 2023-01-24:
I proposed the same as a reaction to that email but it was "ignored" so I don't hope for SDK :-(

maks, 2023-01-26:
@zikajk I expect that being a small team, the developers at Polyend haven't so much "ignored" your email but merely not yet replied to it 🙂 
All I can hope is that the more people express interest and support here, the more chance there is that Polyend would consider this as an option to give the excellent Tracker an even longer and richer life.

PRAYTHEPO, 2023-01-27:
> @zikajk I expect that being a small team, the developers at Polyend haven't so much "ignored" your email but merely not yet replied to it 🙂
>
> All I can hope is that the more people express interest [...]

613 issues in this git. Some bugs exist more than a year. For example LFO and arpeggio clicks.

They don't care. Just buy new silver edition for double price of the original PT and go away. Looks like Todd Howard sold them training courses "How to sell Skyrim ten times in a row"

zikajk, 2023-01-27:
Medusa has got a similar history. Amazing product on paper, but never fully finished software.